microbit-foundation/dev-docs #11: We should explain a little more what slack is

jaustin, 2016-11-02:
Some people may not know what Slack is, and at the moment that's our main engagement system.

We should also be clear we don't consider slack a replacement for a forum - as that's put some people off. Drawing the line that Slack is equivalent to IRC/for live collaboration and other communications means will come in time might help clear this up.

clowrey, 2016-11-14:
Is there a forum type system setup somewhere then? Maybe a https://gitter.im/ message board would work? or is that too much like Slack. 

I too had not heard of Slack before but just signed up to be added to your Slack group.

carlosperate, 2016-11-15:
There is also http://support.microbit.org where people could ask questions and search.

clowrey, 2016-11-15:
Thanks I will try there first. I think I understand why there is no open forum now. The BBC/microbit are trying to strictly control the public spaces around this project as it involves students etc and they don't want uncontrolled dialog with all the variability in appropriateness that goes along with that.. Which is why the online IDEs done have a way to easily post public projects and there is no unrestricted forum for users. - this is probabaly explained somewhere but might be helpful to have as an explanation above your slack link.

microbit-mark, 2018-03-28:
Closed by https://github.com/microbit-foundation/dev-docs/pull/52

closed 2018-03-28 by microbit-mark