microbit-foundation/dev-docs #20: pin numbers on PCB

whaleygeek, 2017-01-10:
(1) On the Edge Connector Data sheet, there is nothing to say what the pin numbers on the PCB are, so it would help to have the reference to the pins (as is available in the here link on the more info page).

microbit.org-ticket: 541

microbit-mark, 2018-03-28:
@whaleygeek Do you have the original link in your email that is referenced by "as is available in the here link on the more info page"? It would be nice to know what the user is refering to.

DavidWhaleMEF, 2018-03-31:
Yes, he means the here link on this page...
[image]

Which goes to this data sheet
[image]

So, rather than having to go all over the place to work out which pin number on the PCB connector connects to P0, P1, GND etc on the actual micro:bit we could add a column to this table...

[image]

DavidWhaleMEF, 2018-03-31:
@microbit-mark I would remove the 'mod' column as that is probably not really used by anyone, and the PCB edge connector is used by lots of people. Space in this table is very limited, so best to prioritise.