microbit-foundation/dev-docs #25: hardware description page acc/mag swapped

whaleygeek, 2017-01-10:
(6) On the hardware description, the part numbers for the accelerometer and magnetometer are swapped.

microbit.org-ticket: 541

04rlowe, 2017-11-04:
I believe this issue has been corrected (?)
[image]

itog, 2018-01-22:
The description on the image is swapped.
Additionally, the accelerometer IC name is not correct.

[image]

microbit-mark, 2018-04-16:
@jaustin Were the SVGs made in Inkscape? I was hoping to update the images, but can't seem to get it to run on my mac. Can't find an alternative that works either.

closed 2018-05-01 by microbit-mark