microbit-foundation/dev-docs #29: hardware/schematic.md doesn't actually have a schematic?

mgrusin, 2017-01-26:
The "Schematics" page (http://tech.microbit.org/hardware/schematic/) doesn't appear to provide an actual schematic for the Microbit. If this is intentional, the page probably shouldn't be called "Schematics." Otherwise, a clearly-labeled link to a PDF of the schematic, along with a link to the source files, would be appreciated. Thank you!

whaleygeek, 2017-02-02:
Yes, sorry about this. It's on the list.

You can get at the schematics for the reference design here:

http://tech.microbit.org/hardware/reference-design/

https://github.com/microbit-foundation/microbit-reference-design/tree/master/PDF/Schematic%20Print

04rlowe, 2017-11-04:
@whaleygeek as per http://tech.microbit.org/hardware/schematic/

> The schematic will be kept in version control in the micro:bit GitHub account. It will be released by the BBC once final sign-off around license terms has completed.

Can this now be updated to provide a link to https://github.com/microbit-foundation/microbit-reference-design/tree/master/PDF/Schematic%20Print ?

jaustin, 2017-11-06:
We should link to the top level GitHub repo for the hardware:
https://github.com/bbcmicrobit/hardware

PR #65 (2018-03-29): Update schematic.md
closes #29
closed 2018-03-29 by microbit-mark