microbit-foundation/dev-docs #37: Broken image link

carlosperate, 2017-04-11:
The link to this image is broken: https://developer.mbed.org/media/uploads/JonnyA/600xNxmicrobit_platform_image_2.png.pagespeed.ic.uXq2rT5rpA.png

Can be found in : http://tech.microbit.org/hardware/schematic/
https://github.com/microbit-foundation/dev-docs/blob/4bbb39bf753d61e8700ca4f3d2e1687e64626fb9/hardware/schematic.md

And: http://tech.microbit.org/hardware/edgeconnector/
https://github.com/microbit-foundation/dev-docs/blob/4bbb39bf753d61e8700ca4f3d2e1687e64626fb9/hardware/edgeconnector.md

04rlowe, 2017-11-04:
I believe this issue has been corrected (?)
[image]

carlosperate, 2017-11-04:
Yes, it was fixed in https://github.com/microbit-foundation/dev-docs/commit/46121b98d18dd1270b32c10868b36ed63840c2f2.

closed 2017-11-04 by carlosperate