microbit-foundation/dev-docs #91: Add info about edge connectors and suppliers

microbit-mark, 2018-10-30:
We should add info and contact details for edge connector manufacturers to https://tech.microbit.org/hardware/edgeconnector/

James-Guo-cylconn, 2018-12-06:
Hi this is James,
How could i put our micro bit connector and company information here:
I m a greenhand.
Hope our Micro bit connector could help everyone a lot .

James Guo
Manager

Yuliang Electronics Industry Group
Dongguan Yuliang Electronics Co., LTD
Add:  Building 103, Sanjiang Industrial Area, Hengli Town,Dongguan,Guangdong,China
Tel:   0086-769-83710858/87062880/82325198  Ext:111
Fax:  0086-769-83710918
Mobile: 008613006803781
Email: james@cylconn.com 
Skype: patrick@dgyuliang.com   
Trademanager: dgyuliang
QQ: 277261874
Wechat:gbb1294495275
https://dgyuliang.en.alibaba.com/
www.dgyuliang.net 
www.linkedin.com/in/dgyuliang/ 
REACH/ RoHS/ CE/ ISO9001/ ISO14001/ UL487698
[Micro Bit-connector_40p 90 degree.pdf](https://github.com/microbit-foundation/dev-docs/files/2651431/Micro.Bit-connector_40p.90.degree.pdf)
[Micro Bit-connector_40p 180 degree.pdf](https://github.com/microbit-foundation/dev-docs/files/2651432/Micro.Bit-connector_40p.180.degree.pdf)
[yulaing electronics logo.pdf](https://github.com/microbit-foundation/dev-docs/files/2651433/yulaing.electronics.logo.pdf)

James-Guo-cylconn, 2018-12-06:
> 
> 
> We should add info and contact details for edge connector manufacturers to https://tech.microbit.org/hardware/edgeconnector/

Thanks a lot Mark,nice day.

James-Guo-cylconn, 2018-12-07:
Mark, i issued some info.
why can't i see it?
Thanks
James

microbit-mark, 2018-12-07:
@James-Guo-cylconn Do you host 
Micro Bit-connector_40p 90 degree.pdf
Micro Bit-connector_40p 180 degree.pdf
anywhere that we could link to? rather than adding them as files to our site, it would be better to link to yours if possible.

James-Guo-cylconn, 2018-12-08:
Yes, here attached our pdf files.
I don't know how to do it.
I have did what i can on your website.
http://www.dgyuliang.net/html/279.html 
http://www.dgyuliang.net/html/280.html 

and i write our link behind them
Please help.
Sincerely
James
 
From: Mark
Date: 2018-12-07 23:17
To: microbit-foundation/dev-docs
CC: James Guo; Mention
Subject: Re: [microbit-foundation/dev-docs] Add info about edge connectors and suppliers (#91)
@James-Guo-cylconn Do you host
Micro Bit-connector_40p 90 degree.pdf
Micro Bit-connector_40p 180 degree.pdf
anywhere that we could link to? rather than adding them as files to our site, it would be better to link to yours if possible.

James-Guo-cylconn, 2018-12-08:
[http://www.dgyuliang.net/html/280.html](url)
[http://www.dgyuliang.net/html/279.html
[Micro Bit-connector_40p 90 degree.pdf](https://github.com/microbit-foundation/dev-docs/files/2659202/Micro.Bit-connector_40p.90.degree.pdf)
[Micro Bit-connector_40p 180 degree.pdf](https://github.com/microbit-foundation/dev-docs/files/2659203/Micro.Bit-connector_40p.180.degree.pdf)

](url)

James-Guo-cylconn, 2018-12-08:
Mark, it is OK now?
Could you please help put the link and file on your website?
I do not know how.
Thanks
James

PR #96 (2018-12-20): add alternative connnector to page
Add a table that renders like this

[image]

closes #91
closed 2018-12-20 by microbit-mark