microbit-foundation/dev-docs #94: Suggestion: Merge edgeconnector and edgeconnector_ds pages

microbit-carlos, 2018-12-04:
Right now finding the https://tech.microbit.org/hardware/edgeconnector_ds/ page is quite difficult, you either need to know the name of the page to find it on google or where the link is in the tech site (since it is not directly accessible from the drop down menu). So for all purposes this page is almost a non-existent to most users.

I would suggest to merge the two pages into one, and to use the word "micro:bit pin out" a bit more in the content and/or titles to try to help with SEO.

microbit-mark, 2019-02-22:
Have asked Phil to update microbit.pinout.xyz https://github.com/Gadgetoid/microbit.pinout.xyz/issues/1

microbit-mark, 2019-02-25:
Have time to work on this now, so I've created a PR as per the original suggestion, pulling in the data from edgeconnecter_ds into edgeconnector and I think this is actually ok ie not too wordy as I thought. I've added tags and updated the title and description to include 'pinout'.

microbit-mark, 2019-03-07:
closed by #111

closed 2019-03-07 by microbit-mark