microbit-foundation/dev-docs #115: Add DAL and Devices Newsletter signup

microbit-mark, 2019-03-15:
The [DAL, Editors and devices mailing list signup](https://microbit.us14.list-manage.com/subscribe?u=e1c30f24b90ff3d70275cfff2&id=25403c7650) is currently a bit hidden away in the KB and we’ve only referred to it as part of the motion sensor change so far. https://support.microbit.org/en/support/solutions/articles/19000087020-micro-bit-motion-sensor-hardware-change

We should update the page here https://tech.microbit.org/get-involved/where-to-find/ with information on this list and a button to signup

microbit-mark, 2019-03-27:
requested review from @jaustin for this

jaustin, 2019-03-28:
Cheers Mark, I like this!
I think perhaps better we link direct to Mailchimp than via eepurl. Also I think we should be consistent on capitalisation: 'DAL, Editors and Devices' seems best

PR #116 (2019-03-29): add dal and devices signup
closes #115
closed 2019-03-29 by microbit-mark