microbit-foundation/dev-docs #125: hardware/schematic.md

rosir, 2019-07-07:
Hello,
I did not find the schema of version 1.5.

microbit-mark, 2019-08-23:
Hi @rosir. Thanks for raising this. This is something we are currently working on and I will update this ticket when we are ready to publish.

microbit-mark, 2020-08-06:
Hi @rosir Thanks for your patience whilst we made the 1.5 Schematics public. We've now been able to publish them via this Pull Request https://github.com/bbcmicrobit/hardware/pull/6

Specifically, you can find the 1.5 Schematic in this file

https://github.com/bbcmicrobit/hardware/pull/6/files#diff-7679b1d01d33deba2f6555dada1db6dd

I'll keep this issue open until we've added them to the tech site

microbit-carlos, 2021-05-05:
The public repo is also linked in the https://tech.microbit.org/hardware/schematic/ page, so we can close this issue.
Thanks @rosir for the report!

closed 2021-05-05 by microbit-carlos