microbit-foundation/dev-docs #128: Papers involving Lancaster University.

jamesadevine, 2019-08-23:
### What children's imagined uses of the BBC micro: bit tells us about designing for their IoT privacy, security and safety

**Project abstract:** Ensuring that young people reap the benefits of the Internet of Things requires proactively attending to the risks they may encounter in entering the world this new technology affords. The e-safety guidelines currently taught in UK schools may not sufficiently prepare children for navigating the risks that come with connected devices. In this paper we describe initial results from the PETRAS project IoT4Kids, exploring the privacy and security implications of children programming the BBC micro:bit, an IoT-ready device designed for children. We report on children's (ages 9-10) likely uses of the micro:bit and discuss their implications, highlighting shortcomings of e-safety education and policy guidelines for such uses.

**Level of research:** Conference paper

**Status:** Complete

**Who is involved:** Lancaster University, PETRAS

**What is the area of research:** Computer Science

**Output:** https://digital-library.theiet.org/content/conferences/10.1049/cp.2018.0015

**Contact:** b.knowles@lancaster.ac.uk, j.finney@lancaster.ac.uk, j.devine@lancaster.ac.uk

-----

### MakeCode and CODAL: intuitive and efficient embedded systems programming for education

**Project abstract:** Historically, embedded systems development has been a specialist skill, requiring knowledge of low-level programming languages, complex compilation toolchains, and specialist hardware, firmware, device drivers and applications. However, it has now become commonplace for a broader range of non-specialists to engage in the making (design and development) of embedded systems - including educators to motivate and excite their students in the classroom. This diversity brings its own set of unique requirements, and the complexities of existing embedded systems development platforms introduce insurmountable barriers to entry.

In this paper we present the motivation, requirements, implementation, and evaluation of a new programming platform that enables novice users to create effective and efficient software for embedded systems. The platform has two major components: (1) Microsoft MakeCode (www.makecode.com), a web app that encapsulates an accessible IDE for microcontrollers; and (2) CODAL, an efficient component-oriented C++ runtime for microcontrollers. We show how MakeCode and CODAL combine to provide an accessible, cross-platform, installation-free, high level programming experience for embedded devices without sacrificing performance and efficiency.
**Level of research:** Conference paper

**Status:** Complete

**Who is involved:** Lancaster University, Microsoft Research

**What is the area of research:** Computer Science

**Output:** https://www.sciencedirect.com/science/article/pii/S1383762118306088

**Contact:** j.finney@lancaster.ac.uk, j.devine@lancaster.ac.uk, tball@microsoft.com

----

### Energy in Schools: Promoting Global Change through Social Technical Deployments

**Project abstract:** Reducing carbon emissions is a key priority across the globe, and in the UK, schools have been identified as the second largest users of non-domestic energy. In this paper, we present an IoT solution for schools that aims to unite senior leadership, teachers, and pupils in the goal of reducing or shifting their energy consumption and carbon emissions. We achieve this by prompting behavioural change through instrumenting schools with sensors, visual displays, and a variety of educational resources which use the BBC micro:bit to interact with the data produced by these sensors, enabling pupils to engage in educational activities to solve real world problems. By increasing the visibility, availability, and interactivity of data, we enable a new space for dialogue between facilities managers and building users. We summarise some of the challenges and lessons learned so far, with preliminary results indicating our approach is effective in raising the profile of energy management and shifting demand. Future monitoring and evaluation will provide more detail on the effectiveness of our IoT solution.

**Level of research:** Conference paper

**Status:** Complete

**Who is involved:** Lancaster University, CSE, Samsung

**What is the area of research:** Computer Science

**Output:** https://eprints.lancs.ac.uk/id/eprint/133635/

**Contact:** j.finney@lancaster.ac.uk, j.devine@lancaster.ac.uk, k.new1@lancaster.ac.uk

----

### The BBC micro:bit––from the UK to the World

**Project abstract:** The micro:bit is a small programmable and embeddable computer designed, developed, and deployed by the BBC and 29 project partners to approximately 800,000 UK Year 7 (11/12 year old) school children in 2015-2016. The micro:bit has been described by the BBC as "its most ambitious education initiative in 30 years, with an ambition to inspire digital creativity and develop a new generation of tech pioneers". This article describes the design of the BBC micro:bit, the realization of its goals through examples of the diverse projects crested with the micro:bit, reviews the project's history as it transitioned from a UK-centric to a worldwide project, and concludes with lessons learned and project outcomes.

**Level of research:** Conference paper

**Status:** Complete

**Who is involved:** Lancaster University, micro:bit foundation, BBC.

**What is the area of research:** Computer Science

**Output:** https://eprints.lancs.ac.uk/id/eprint/135065/1/cacmMicrobitRevisted2019_1_31.pdf

**Contact:** j.finney@lancaster.ac.uk, j.devine@lancaster.ac.uk

----

### Physical computing: A key element of modern computer science education

**Project abstract:** Policymakers and educators around the globe acknowledge the importance of computer science (CS) education. But traditional CS teaching tools and methodologies do not necessarily address the needs of a diverse, global student population or the latest developments in modern programming and data science. Physical computing – combining software and hardware to build interactive physical systems that sense and respond to the real world – has been shown to result in broad engagement across a spectrum of users. In this paper we review prior research into physical computing in the classroom and combine this with our own experiences. We summarise the reported benefits and show how recent trends in the design and implementation of physical computing devices and systems are resulting in growing adoption. By way of example, we provide a detailed description of a recently developed physical computing system, the BBC micro:bit.

**Level of research:** Conference paper

**Status:** Complete

**Who is involved:** Lancaster University, Microsoft Research, Kings College London.

**What is the area of research:** Computer Science

**Output:** https://eprints.lancs.ac.uk/id/eprint/135064/

**Contact:** j.finney@lancaster.ac.uk, steve.hodges@microsoft.com

----

microbit-mark, 2019-08-23:
Brilliant. Thanks for the submissions @jamesadevine

microbit-mark, 2019-09-09:
Now released in https://tech.microbit.org/labs/ 🎉 

Thanks again @jamesadevine

closed 2019-09-09 by microbit-mark