microbit-foundation/dev-docs #131: Mention of configuring high drive mode in article about connecting an LED

martinwork, 2019-12-04:
Not sure this is an error... Having read the discussion at
https://github.com/lancaster-university/microbit-dal/issues/456
where it was suggested that "high drive" was disabled and that it is not set anywhere in the DAL, I noticed...
https://support.microbit.org/support/solutions/articles/19000101863-connecting-an-led-to-the-micro-bit
"Each pin of the micro:bit also has a current supply limit of 5mA and a maximum of 3 pins can be configured in this high drive mode at one time.".

Also, the link in that text leads to https://hardsoft-support.kayako.com/conversation/new.

microbit-mark, 2019-12-06:
Thanks @martinwork. This statement is taken from https://tech.microbit.org/hardware/#general-purpose-inputoutput-pins
https://github.com/microbit-foundation/dev-docs/commit/1d2d2d8e52197f52bfe188e68030788f59974106#diff-206e06b1daef089df342a11027650cb3R208

which references the nrf51 data sheet
[nRF51822_PS_v3.1.pdf](https://github.com/microbit-foundation/dev-docs/files/3931495/nRF51822_PS_v3.1.pdf)

In Section 8.23 they note
"VOH Output high voltage (high-drive, 5 mA). 1 VDD-0.3 VDD V"
Maximum number of pins with 5 mA high drive is 3.

But perhaps this information isn't relevant if the micro:bit nrf51 is pre-configured with this mode disabled?

martinwork, 2019-12-06:
I'm sorry, I don't know. The discussion at lancaster-university/microbit-dal#456 led me to look at this page. I am reading in the hope of learning something! Hopefully that discussion will shed some light. The latest comment suggests to me that all is well except the reference to high drive, and maybe the 3 pin limit, may not be the appropriate wording.