microbit-foundation/dev-docs #153: GitHub action result is "pass" even if it finds errors in the URLs

microbit-carlos, 2020-08-14:
I assume the main reason the link checker step is failed is so that the next step (posting a comment) runs.

Is there a way to fail the action and still have the github issue automatically opened?

microbit-mark, 2020-08-18:
I think you can fail the action if it finds errors, https://github.com/peter-evans/link-checker#creating-a-failing-check-for-link-errors, but it defaults to passing and creates the report. Is it useful to have the action fail as well as generating the issue?

microbit-carlos, 2020-08-18:
It's probably fine as it is, it's just a bit odd when there are issues and looking at the actions you see all of them marked as "pass".
If it's very simple to mark it as failed and still have the CI add the GH issue, i'd say is worth the fix, but it's more than a quick look-up the docs and a test, then it's not worth the effort.

closed 2020-12-16 by microbit-mark