microbit-foundation/dev-docs #276: A11y: Spec pages have two h1 tags

microbit-mark, 2021-02-08:
## Description
A clear and concise description of what the issue is

Web pages should only have 1 h1 tag for screenreader accessibility

Adding the specs as submodules means that they have a header in the README and a header in the frontmatter

[image]

[image]

We could update the front matter to remove the title, but it makes the pages inconsistent with the rest of the site. 

## Expected behaviour
We should only have one h1 heading per page

## Screenshots
If applicable, add screenshots to help explain the issue

StarTrooper08, 2022-04-06:
Can I work on this issue. I m new to Microbit community.

microbit-matt-hillsdon, 2026-10-12:
Resolved by (or perhaps before) the new site. For each page:

[image]

closed 2026-10-12 by microbit-matt-hillsdon