microbit-foundation/dev-docs #354: software/interface.md: document that macos 11+ requires config for blhost to work.

jaustin, 2021-12-09:
Without following the instructions at :
https://community.nxp.com/t5/LPC-Microcontrollers/MCU-Link-blhost-issue-on-macOS/m-p/1193204 (ie allowing terminal to monitor input in the macOS security preference pane)

I will always get the following error when trying to use blhost

Error: UsbHidPeripheral() cannot open USB HID device

I'm not sure if there are more appropriate workarounds for this, so not making an edit right now but @carlosperate did you have to do this?

carlosperate, 2021-12-10:
I didn't have to manually do this when using blhost, but that's very likely because my environment was already set up like this:
[image]

I'm pretty sure I have not **manually** added the terminal apps this this "Input Monitoring" service (I have done for others though), so I must have been prompt by the OS when running this or some other terminal app.

microbit-carlos, 2022-01-12:
@jaustin which version of blhost do you have?
I wonder if the latest 2.6.7 release (I've been personally been using 2.6.6) could have fixed this: https://www.nxp.com/design/software/development-software/mcuxpresso-software-and-tools-/mcu-bootloader-for-nxp-microcontrollers:MCUBOOT?&tab=Design_Tools_Tab