microbit-foundation/dev-docs #389: Reinstate Cylconn/Yuliang connector

jaustin, 2022-12-02:
## Description

In #96 we added some useful connectors to the tech site.

[image]

I think in subsequent refactor these got lost. 

There is now also http://www.dgyuliang.net/html/282.html which is even more micro:bit specific.

@microbit-robert could you take a quick look please at when we lost the connector examples, whether there was a specific reason to do it, and if not let's put these back in.

## Expected behaviour

Cylconn/Yuliang (same company as far as I understand) connectors are reinstated

microbit-robert, 2022-12-02:
@jaustin It looks like the links were removed because they weren't resolving. 

https://github.com/microbit-foundation/dev-docs/commit/15d97b805026cf65b018791d091944a2a1bf3b62

The links resolve just fine now, so we can re-instate the old links, or replace them with new links for the 90 and 180 connectors as noted above. Do you have a preference?

jaustin, 2022-12-02:
I don't really, but guess this is probably the most useful page? http://www.dgyuliang.net/Produtcs/Customized%20Connector/

I notice it's not https which might be what caused errors. There's also this https://dgyuliang.en.alibaba.com/productgrouplist-807517501/edge_connector_for_BBC_micro_bit.html?spm=a2700.shop_index.88.8

I suggest we link to both, something like

```|[Cyclonn/Dongguan Yuliang](http://www.dgyuliang.net) | [micro:bit connectors](http://www.dgyuliang.net/Produtcs/Customized%20Connector/), also see [Alibaba store](https://dgyuliang.en.alibaba.com/productgrouplist-807517501/edge_connector_for_BBC_micro_bit.html?spm=a2700.shop_index.88.8)|```
```

closed 2022-12-09 by microbit-robert